freemint/fvdi #4: Font Rendering Performance...

mfro0, 2020-05-18:
... is - ehm - dissatisfactory.

Even on the Firebee, it takes about 5 seconds to redraw a 640x480 sized window with about 40 different TrueType fonts. On hatari (32 MHz 040 TT emulation) the same window redraw takes > 25 seconds. No improvement after initial display (i.e. this is for every redraw).

Font cache settings don't seem to make any difference (other than crashes due to lack of memory elsewhere when too large). Is the font caching doing anything other than using up memory? Don't think so.

For comparision: NVDI (5.02) on hatari renders the same screen near instantly (80 ms) once initial font caching has been completed.

th-otto, 2020-05-18:
Font caching (if any) is done by freetype. I think it can be tweaked by applications, but i don't know how to achieve that from a VDI interface (that is usually done by calling some property setting functions on the faces, and/or drivers).

You can also try different freetype versions. The ones i listed on the ML (2.2.1, 2.5.2, 2.8.1 and 2.10.2) should work at least.

But generally, it will be hard to achieve any performance that comes close to NVDI, or even SpeedoGDOS. I do my current tests with a version without freetype support at all, because loading the fonts alone takes ~1min on a 32Mhz/TT emulation.

mfro0, 2020-05-18:
I have around 50 fonts installed on my Firebee and initial loading takes less than a second. It's the font rendering that is awfully slow. If I test NVDI (on hatari), the initial open of the font window takes maybe 10-20s+ as well, but afterwards it's really fast.

With fVDI, it appears the font information is reread from disk on each window redraw. More exact, at each vqt_extent() call, it's not even the drawing that causes the slowdown as I tried to suppress that outside the clipping window which didn't really improve redraw time (it takes about the same time if it redraws an 80x160 window or a fullscreen one).

There is certainly a lot of room for performance improvement.

mfro0, 2020-05-18:
increasing MiNT's disk cache helps a lot. Redraw can be made like 5xfaster (while setting the font cache in fvdi.sys doesn't seem to do anything).

th-otto, 2020-05-19:
> initial loading takes less than a second.

For me that already takes way too long (using Hatari/TOS 3.06/32Mhz TT, so there is no mint disk cache). The first number in the list below is the size of the font file, the last one time in seconds (obtained from hz_200) to load it:

```
   Load font:   774476 C:\gemsys\ttf\ARIAL.TTF 0.150
   Load font:   563800 C:\gemsys\ttf\ARIAL_BI.TTF 0.125
   Load font:   751104 C:\gemsys\ttf\ARIAL_BO.TTF 0.155
   Load font:   557760 C:\gemsys\ttf\ARIAL_IT.TTF 0.125
   Load font:    22400 C:\gemsys\ttf\ATSYS.TTF 0.025
   Load font:    22788 C:\gemsys\ttf\ATSYSB.TTF 0.025
   Load font:    25372 C:\gemsys\ttf\ATSYSBI.TTF 0.025
   Load font:    25108 C:\gemsys\ttf\ATSYSI.TTF 0.025
   Load font:    53456 C:\gemsys\ttf\BRUSH_SC.TTF 0.035
   Load font:    95828 C:\gemsys\ttf\CENT_B.TTF 0.045
   Load font:    90948 C:\gemsys\ttf\CENT_BI.TTF 0.040
   Load font:    92120 C:\gemsys\ttf\CENT_I.TTF 0.045
   Load font:    97004 C:\gemsys\ttf\CENT_R.TTF 0.205
   Load font:   120120 C:\gemsys\ttf\COMIC_B.TTF 0.045
   Load font:   135484 C:\gemsys\ttf\COMIC_R.TTF 0.045
   Load font:   691796 C:\gemsys\ttf\COUR_B.TTF 12.865
   Load font:   506640 C:\gemsys\ttf\COUR_BI.TTF 6.560
   Load font:   589900 C:\gemsys\ttf\COUR_I.TTF 0.065
   Load font:   684624 C:\gemsys\ttf\COURIER.TTF 0.105
   Load font:   568656 C:\gemsys\ttf\DEJAVULG.TTF 0.170
   Load font:   298928 C:\gemsys\ttf\DEJAVULG.TTF 0.175
   Load font:   247260 C:\gemsys\ttf\DEJAVULG.TTF 0.170
   Load font:   117072 C:\gemsys\ttf\DROIDSAN.TTF 0.035
   Load font:   298928 C:\gemsys\ttf\DVMONO.TTF 0.065
   Load font:   154644 C:\gemsys\ttf\DVSAN.TTF 0.235
   Load font:   138656 C:\gemsys\ttf\DVSANB.TTF 0.065
   Load font:   148120 C:\gemsys\ttf\DVSANBO.TTF 0.075
   Load font:    97452 C:\gemsys\ttf\DVSANM.TTF 0.035
   Load font:    89828 C:\gemsys\ttf\DVSANMB.TTF 0.035
   Load font:    97416 C:\gemsys\ttf\DVSANMBO.TTF 2.480
   Load font:   103832 C:\gemsys\ttf\DVSANMO.TTF 1.945
   Load font:   140716 C:\gemsys\ttf\DVSANO.TTF 0.060
   Load font:   568656 C:\gemsys\ttf\DVSANS.TTF 6.635
   Load font:   131696 C:\gemsys\ttf\DVSER.TTF 2.460
   Load font:   130796 C:\gemsys\ttf\DVSERB.TTF 4.765
   Load font:   119088 C:\gemsys\ttf\DVSERBO.TTF 2.250
   Load font:   247260 C:\gemsys\ttf\DVSERIF.TTF 0.235
   Load font:   118012 C:\gemsys\ttf\DVSERO.TTF 0.060
   Load font:   186000 C:\gemsys\ttf\GARA_B.TTF 0.060
   Load font:   175344 C:\gemsys\ttf\GARA_BI.TTF 2.170
   Load font:   175508 C:\gemsys\ttf\GARA_I.TTF 2.290
   Load font:   185912 C:\gemsys\ttf\GARA_R.TTF 4.160
   Load font:    67416 C:\gemsys\ttf\GILLSA.TTF 0.040
   Load font:    68304 C:\gemsys\ttf\GILLSAB.TTF 0.040
   Load font:    71496 C:\gemsys\ttf\GILLSABI.TTF 0.040
   Load font:    69436 C:\gemsys\ttf\GILLSAI.TTF 2.490
   Load font:   137000 C:\gemsys\ttf\LETGOT_B.TTF 2.195
   Load font:   129924 C:\gemsys\ttf\LETGOT_I.TTF 2.090
   Load font:   129580 C:\gemsys\ttf\LETGOT_R.TTF 0.050
   Load font:   179492 C:\gemsys\ttf\NARROW.TTF 2.315
   Load font:   184420 C:\gemsys\ttf\NARROWB.TTF 2.300
   Load font:   183932 C:\gemsys\ttf\NARROWBI.TTF 7.165
   Load font:   184944 C:\gemsys\ttf\NARROWI.TTF 4.670
   Load font:   416488 C:\gemsys\ttf\PALAT_B.TTF 5.735
   Load font:   332912 C:\gemsys\ttf\PALAT_BI.TTF 3.730
   Load font:   410260 C:\gemsys\ttf\PALAT_I.TTF 2.175
   Load font:   469100 C:\gemsys\ttf\PALATINO.TTF 1.155
   Load font:   836380 C:\gemsys\ttf\TNR.TTF 1.385
   Load font:   842964 C:\gemsys\ttf\TNR_B.TTF 0.155
   Load font:   622232 C:\gemsys\ttf\TNR_BI.TTF 0.125
   Load font:   665224 C:\gemsys\ttf\TNR_I.TTF 0.125
   Load font:    49052 C:\gemsys\ttf\VERAMOBD.TTF 0.030
   Load font:    55032 C:\gemsys\ttf\VERAMOBI.TTF 0.030
   Load font:    54508 C:\gemsys\ttf\VERAMOIT.TTF 0.030
   Load font:    49224 C:\gemsys\ttf\VERAMONO.TTF 0.030
   Load font:   156340 C:\gemsys\ttf\VERDA_B.TTF 0.060
   Load font:   176528 C:\gemsys\ttf\VERDA_BI.TTF 0.200
   Load font:   177996 C:\gemsys\ttf\VERDA_I.TTF 0.065
   Load font:   189144 C:\gemsys\ttf\VERDANA.TTF 0.065
   Load font:   124308 C:\gemsys\ttf\WEBDINGS.TTF 0.030
   Load font:    83496 C:\gemsys\ttf\WINGDIN1.TTF 5.065
   Load font:    65264 C:\gemsys\ttf\WINGDIN2.TTF 2.550
   Load font:    34804 C:\gemsys\ttf\WINGDIN3.TTF 2.590
   Load font total: 100.115
```

Especially i'm a bit buffled why times are so different. For example, for courier it takes much longer to load the bold variant than the regular one, but for Garamont it is the other way around.

Test were done using freetype 2.2.1, but freetype 2.5.2 gives almost the same results.

mfro0, 2020-05-21:
Did you realize that your latest commits broke m68k (non-Coldfire) builds?

Am 19.05.20 um 09:10 schrieb Thorsten Otto:

th-otto, 2020-05-22:
What  do you mean? It was supposed to fix this info:

[image]

mfro0, 2020-05-22:
see #5  ...

th-otto, 2020-05-22:
Ah ok. Didn't get a email notification about that new issue...

closed 2022-04-27 by mfro0