freemint/fvdi #5: commit 7d17626 breaks build

mfro0, 2020-05-21:
fVDI does not build for me with commit 7d17626 ("We do not need to preprocess some simple assembler include files")

th-otto, 2020-05-22:
Works fine here. However two of the files are symlinks, can that cause problems?

th-otto, 2020-05-22:
Hm i don't get it. Just tried in my local fork, and in the snapshot it get
```
$ ls -l ../include
ls -l ../include
total 112
-rw-rw-r-- 1 travis travis  1233 May 22 07:29 calamus.h
-rw-rw-r-- 1 travis travis  9459 May 22 07:29 function.h
-rw-rw-r-- 1 travis travis 18067 May 22 07:29 fvdi.h
-rw-rw-r-- 1 travis travis  2605 May 22 07:29 globals.h
-rw-rw-r-- 1 travis travis  2236 May 22 07:29 macros.dev
-rw-rw-r-- 1 travis travis  1386 May 22 07:29 macros.gnu
-rw-rw-r-- 1 travis travis   126 May 22 07:29 macros.inc
-rw-rw-r-- 1 travis travis  1565 May 22 07:29 macros.tas
drwxrwxr-x 2 travis travis  4096 May 22 07:29 modules
-rw-rw-r-- 1 travis travis  6839 May 22 07:29 nf_ops.h
-rw-rw-r-- 1 travis travis   529 May 22 07:29 os.h
-rw-rw-r-- 1 travis travis  2685 May 22 07:29 relocate.h
drwxrwxr-x 2 travis travis  4096 May 22 07:29 stdio
drwxrwxr-x 2 travis travis  4096 May 22 07:29 stdlib
drwxrwxr-x 2 travis travis  4096 May 22 07:29 string
-rw-rw-r-- 1 travis travis  9959 May 22 07:29 types.inc
-rw-rw-r-- 1 travis travis  4449 May 22 07:29 utility.h
-rw-rw-r-- 1 travis travis   479 May 22 07:29 vdi.inc
```
That is, include/vdi.gnu (the symlink) is not there.

However, when i do a fresh clone, it is...

mfro0, 2020-05-22:
the links are there, that does not seem to be the problem.

You changed the `$(AS)` commands to `$(CC) -x assembler-with-cpp` (why?), but left the `-I...` option. That way, the include path option will be eaten by the preprocessor (that can't do much with it because inside the assembler files we still have the `as` `.include` and no `#include` option).

The `ASFLAGS` include variables must be prepended with `-Wa,-I...`then.

I wonder how this could work for you?

th-otto, 2020-05-22:
Yes, that's the strange thing. My compiler driver (also gcc 4.6.4) passes along the -I directives when invoking as. But Vincents version does not. Should be fixed now.

BTW i wasn't notified by travis that the build failed, otherwise i would have noticed it earlier. Do you remember which setting is responsible for that?

th-otto, 2020-05-22:
The idea for using gcc with  -x assembler-with-cpp was to simplify the conv2gas step, by using c-style comments (as supports that, even when invoked directly, and pasm too). However it turned out that pacf complains about that, so i had to revert the changes to the include files.

mfro0, 2020-05-22:
> My compiler driver (also gcc 4.6.4) passes along the -I directives when invoking as. But Vincents version does not. Should be fixed now.

That seems odd. If your driver really does that, I would consider this a bug, as this will probably make problems more often than not (I cannot think of a situation where it would be the assembler would be desireable).

It seems the travis build notifications are somehow disturbed - the last build obviously succeded but the button on the same page says "failed".
I have just added an explicit build notification but at the moment, there is no build triggered at all?

I'm going to close this issue as the problem is fixed.

closed 2020-05-22 by mfro0

th-otto, 2020-05-22:
> That seems odd.

Yes, that might be true. But that's the reason why i didn't notice the problem.

Maybe i should revert that attempt completely, and directly use AS again. But that whole business about converting the assembler sources drives me mad. There are some dangerous substitutions made there (ie it substitutes all branches to .loop1 labels by 1b etc, if you ever use that in some other order then you are doomed)