freemint/fvdi #6: does not run on FireBee anymore

mfro0, 2020-06-27:
commit de376924d2d602df999e507def8cb442b330c5e6 breaks fVDI for the FireBee.

This commit changed the code in `check_cookies()` to set `cpu` to 0 if the `_CPU` cookie is not set (when `get_cookie()` returned -1).

This is wrong for ColdFire. EmuTOS on the FireBee does not set the `_CPU` cookie at all (but sets the _MCH cookie instead), FireTOS sets it (AFAIK) as if it had a 040 CPU.

Was this done on purpose (I think I have tested that fVDI worked fine in 000 emulation in Hatari?), i.e. is this required or just "to keep things clean"?

If the latter, this should be reverted, in my opinion. If this setting is however required for some reason, I most likely have to revisit all the cases where `cpu` is used and fix them for ColdFire with conditional compilation.

DavidGZ, 2020-06-28:
> 
> This is wrong for ColdFire. EmuTOS on the FireBee does not set the `_CPU` cookie at all (but sets the _MCH cookie instead), FireTOS sets it (AFAIK) as if it had a 040 CPU.

Only to remark that FireTOS sets the _CPU value as a 060

mfro0, 2021-01-01:
fixed with latest commits

closed 2021-01-01 by mfro0