freemint/fvdi #13: CI broken

mfro0, 2022-01-29:
The CI workflow is broken. I'm not familiar with the scripts, but it appears older packages at savannah.gnu.org regularly get moved into the freetype-old directory?

As it stands, one of the packages required for the CI process just moved and isn't found anymore, causing the CI process to bail out prematurely.

@th-otto : is that something you could look into (if time permits), please?

th-otto, 2022-01-29:
On Samstag, 29. Januar 2022 13:56:49 CET Markus wrote:
> @th-otto : is that something you could look into (if time permits), please?

Should be done. Have to check once built is complete.

Should that happen again: it's just a one-liner in one of the install-scripts.

mfro0, 2022-01-29:
I didn't want to fiddle with a script I don't know, but if it's really just a one-liner I can do that next time myself.
Thank you!

closed 2022-01-29 by mfro0