freemint/fvdi #25: Fonts do not sit correctly on baseline

chrisridd, 2022-02-28:
I tried to draw a single line of text in multiple fonts, aligned at the baseline. The results are not great, using 40pt text just to make it a bit more obvious:

[image]

The red line is my baseline. I'm doing:

	vst_alignment(handle, 0, 0, &dummy, &dummy); // left, baseline

	vst_font(handle, fontid1);
	vst_point(handle, 40, &dummy, &dummy, &dummy, &dummy);
	v_ftext(handle, xstart, ystart, "DejaVu");

	vst_font(handle, fontid2);
	vst_point(handle, 40, &dummy, &dummy, &dummy, &dummy);
	xstart += (width-of-DejaVu-text);
	v_ftext(handle, xstart, ystart, "Bookerly");

	vst_font(handle, fontid3);
	vst_point(handle, 40, &dummy, &dummy, &dummy, &dummy);
	xstart += (width-of-Bookerly-text);
	v_ftext(handle, xstart, ystart, "Helvetica");

NVDI 5 does get the vertical alignment correct.

The `ft2_text_render()` assembles the character bitmaps into a bigger buffer and blits it to the screen using a vertical offset from `extra.distance`. The computation of the `extra.distance` fields **seems** to be OK, but the more I read about metrics and sizes at freetype.org the less I feel I know! Is the bbox the right way to measure things, or just the ascent+descent?

The errors are reasonably significant (5 pixels in my screenshot), so they can't be just simple rounding errors.

chrisridd, 2022-03-01:
After the patch in #26 this is what I get:

[image]

mfro0, 2022-03-02:
Looks better now for me as well. Thank you for your contribution!

closed 2022-03-02 by mfro0