freemint/fvdi #40: Teradesk Preferences dialog redraw errors in 8bpp

mikrosk, 2022-06-11:
It seems that there is a redraw error in 8bpp resolutions: freemint/freemint#256.

xdelatour, 2023-04-29:
- No bug with Hatari + EmuTOS + fvdi + bitplane.sys (Falcon, 256c)
- No redraw error with ARAnyM + fvdi + aranym.sys when hardware blitting is disabled:
`01r aranym.sys mode 1024x768x8@60 irq accelerate xxxx-xx`
On the other hand, without hardware acceleration, bitmaps are totally destroyed in 8bpp mode and have wrong colors in 24bpp and 32bpp

mikrosk, 2023-04-29:
So the original bug is still present when hardware blitting is _enabled_, right?

xdelatour, 2023-04-29:
Yes, you're right:
- Redraw error with hardware blitting enabled [freemint#256](https://github.com/freemint/freemint/issues/256#issue-1260667932)
- Blitting error with hardware blitting disabled (but no problem with redraw after closing the dialog box)
[image]