freemint/fvdi #41: FVDI building for 680x0 depends on installed PortAsm

ctirad, 2022-06-14:
Please, remove PortAsm dependency for 68k.

cdpjenkins, 2022-06-14:
I think I hit this problem a while back when I first built fvdi (and I didn't do anything about it - sorry!). PortAsm isn't even available on the platforms that I use (M1 Mac and ARM Linux).,

The problem that I hit is that the buildit.sh script always builds drivers/radeon, and drivers/radeon/Makefile always sets `CPU     = v4e`. A simple fix would be to only build radeon if the CPU is set to v4e. I'd be happy to raise a PR to do that when I get some time. How does that sound?

cdpjenkins, 2022-06-14:
FYI the hack that I added to get it to build for me on my Mac:

```
index ad3b2bf..2ae9972 100755
--- a/buildit.sh
+++ b/buildit.sh
@@ -1,5 +1,7 @@
 #!/bin/bash
 
+set -ex
+
 CPU=${CPU:-"020"}                      # assign defaults if not set
 DEBUG=${DEBUG:-"yes"}
 
@@ -30,15 +32,15 @@ rm -f fvdi_gnu.prg
 (cd ${BPL_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make clean) || exit
 (cd ${BPL_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make -j) || exit
 
-(cd ${RAD_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make clean) || exit
-(cd ${RAD_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make -j) || exit
+# (cd ${RAD_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make clean) || exit
+# (cd ${RAD_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make -j) || exit
 
 (cd ${FBE_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make clean) || exit
 (cd ${FBE_DIR}; DEBUG=${DEBUG} CPU=${CPU} M68K_ATARI_MINT_CROSS=yes make -j) || exit
 
 cp ${BPL_DIR}/bitplane.sys .
 cp ${ENG_DIR}/fvdi_gnu.prg .
-cp ${RAD_DIR}/radeon.sys .
+# cp ${RAD_DIR}/radeon.sys .
 cp ${FBE_DIR}/firebee.sys .
 
 #m68k-atari-mint-strip bitplane.sys
diff --git a/fvdi/drivers/radeon/Makefile b/fvdi/drivers/radeon/Makefile
index 93e8937..0167ce4 100644
--- a/fvdi/drivers/radeon/Makefile
+++ b/fvdi/drivers/radeon/Makefile
@@ -14,7 +14,7 @@ all: $(TARGET)
 
 top_srcdir = ../..
 
-CPU     = v4e
+# CPU     = v4e
 OPTS    = -mshort -O2 -fomit-frame-pointer
 include $(top_srcdir)/CONFIGVARS
```

I'm not quite sure how much of this is needed but I didn't spend much time on it... I just wanted the build to work!

mfro0, 2022-06-14:
I would just recommend to get rid of `buildit.sh` alltogether. It was meant to just help myself in a very early stage when we took over fVDI development and the Makefiles weren't doing what they are supposed to.

cdpjenkins, 2022-06-14:
Ah nice, I was wondering why we have both buildit.sh *and* a top level makefile. It looks like the CI calls the top level makefile too.

So if I get the time too raise a PR later on, it'll be a PR that deletes buildit.sh (and possibly changes the docs, which I think refer to it).

cdpjenkins, 2022-06-14:
I raised #42 to remove buildit.sh.

I've tested building the 000 and 020 versions on my Mac M1. (Can't test v4e because no PortAsm :-). Hopefully the CI should do a good job of confirming that it builds on Linux as well.

closed 2022-07-07 by mfro0