freemint/fvdi #45: Problem with FVDI+NVDI vr_transfer_blitz()

AtariTrek, 2023-02-17:
"fvdi + nvdi" doesn't report vr_transfer_bits() is supported
vr_transfer_bits() seems incomplete

It makes Lonny Pursells program "imview" not detect vr_transfer_bits() correctly.
And vs_ctab_entry() commands are unsupported, which makes fvdi uncomplete compared to nvdi.

Thank you.

mikrosk, 2023-10-18:
@AtariTrek can you provide more details? I.e. how do you run fvdi and nvdi (auto, mint.cnf, what order, what machine) and some code snippet to test?

AtariTrek, 2023-10-24:
Hi Miro!
The program imgview (an excellent program) crash when "fulling" the screen
image while using fvdi.
While using plain nvdi it doesn't.
https://atari.gfabasic.net/
This is due to the fact that a NULL address with vr_transfer_bits() makes
fvdi crash, while the nvdi docs clearly states that a Null address can be
used.
I hope this helps ;)
Thank you

Den ons 18 okt. 2023 kl 20:25 skrev Miro Kropáček ***@***.***

AtariTrek, 2023-10-24:
> :
> Ah ok, so it's "just" a plain fVDI bug. I was confused with the
> combination with NVDI.

Well both yes and no.
It is with the combination fvdi+nvdi that makes vr_transfer_bits() crash
together with dither_bits
thank you

Den tis 24 okt. 2023 kl 11:11 skrev Miro Kropáček ***@***.***

mikrosk, 2023-10-24:
Can you please test just fvdi alone then?

AtariTrek, 2023-10-24:
Dither_bits does not exist with fvdi alone, it is needed with nvdi.
If you install the program I provided (imgview+modules) you will be able to
replicate the problem.
The other modes perhaps does the same thing with vr_transfer_bits(), I am
not sure.
I use aranym windows and do not have fvdi installed at the moment.
Just nvdi and blowup.

Den tis 24 okt. 2023 kl 15:07 skrev Miro Kropáček ***@***.***

AtariTrek, 2023-10-25:
I have now installed fvdi, but it shouldn't matter for test purposes since
I can not test vr_transfer_bits(/) without dithering and nvdi 5.03?

AtariTrek, 2023-10-25:
The problem would most surely be fvdi since the null adress works with
nvdi only without problems.