freemint/fvdi #47: Drawing bug

xdelatour, 2023-04-10:
While doing some tests i found this:

[image]

I don't know where the problem comes from. Is it caused by a single function or by the combination of several?

```
	wind_get (hwind, WF_WORKXYWH, &xw, &yw, &ww, &hw);	/* Zone de travail */
	pxy[0] = xw;                				/* Preparer effacement fenetre */
	pxy[1] = yw;
	pxy[2] = xw + ww - 1;
	pxy[3] = yw + hw - 1;
	vswr_mode (handle, MD_REPLACE);  		 	/* Dessin en mode Remplacement */
	vsf_color (handle, 0);            			/* Couleur blanche */
	v_bar (handle, pxy);              			/* "Vider" la fenetre */
	vsf_color (handle, 1);					/* Couleur noire */
	vsf_perimeter (handle, 1);				/* Tracer le contour */

	...

	case T_ELL :						/* Ellipses */
		vsf_interior (handle, FIS_HOLLOW);		/* Vide */
		vsf_perimeter(handle, 1);
		vswr_mode (handle, MD_TRANS);			/* Mode transparent */
		for (i = 0 ; i < hw ; i += 3)			/* Boucle de dessin */
			v_ellipse (handle, xw + (ww / 2), yw + (hw / 2), i , (hw / 2) - i); /* Ellipse */
		break;
```

xdelatour, 2023-04-10:
Maybe related to #34 ?

Atari260STE, 2024-01-14:
the same story with DRI APSKEL.C exaple, v_ellipse isn't filled with the pattern:
```
/****************************************************************/
/* Draw Filled Ellipse.						*/
/****************************************************************/
draw_sample()
{
int temp[4];
	vsf_interior(handle,2);
	vsf_style(handle,8);
	vsf_color(handle,0);
	temp[0]=xwork;
	temp[1]=ywork;
	temp[2]=xwork+wwork-1;
	temp[3]=ywork+hwork-1;
	v_bar(handle,temp);		/* blank the interior */
	vsf_interior(handle,4);
	vsf_color(handle,1);
	v_ellipse(handle,xwork+wwork/2,ywork+hwork/2,wwork/2,hwork/2);
}
```
[image]

Atari260STE, 2024-01-14:
moreover, APSKEL's window is completely empty in case of fvdi-0.96a-ft2.8.1-c19-000.zip