freemint/fvdi #49: fvdi doesn't compile with m68k-atari-mintelf toolchain

mikrosk, 2023-10-18:
It has m68k-atari-mint-gcc hardwired on several places.

th-otto, 2024-12-26:
I think this cannot be fixed easily. First off, all assembler files are preprocessed, and a lot of them translated by pacf for firebee. Using elf toolchains there may give surprising results. Another problem is that using elf, you will get clashes with some symbols, because for example `_handle` (a global variable) will then become `handle`, but that same name is also used by vdi.inc, for an offset into the control array.

mikrosk, 2025-02-08:
I wonder what was the idea behind https://github.com/freemint/fvdi/commit/c05474eaa3627cd27a6e114df079151899f59baa  then. Having at least those `-latest` suffixes working would be a nice start for https://github.com/freemint/freemint/issues/322.