freemint/fvdi #53: Drop FT2 2.5.x, 2.8.x, 2.10.x builds

mikrosk, 2025-02-08:
According to https://github.com/freemint/fvdi/commit/9d03562eb1f2790db197ca72a60dbe5870700e8e:

fvdi_gnu.prg sizes with gcc10 are:

| Version | Binary |
| ------- | ------ |
| 2.2.1   | 413489 |
| 2.5.2   | 552275 |
| 2.8.1   | 544862 |
| 2.10.2  | 571212 |
| 2.10.3  | 572855 |
| 2.10.4  | 572851 |
| 2.11.0  | 577312 |
| 2.11.1  | 572461 |

They are not only confusing but also perhaps also not that much needed: keeping 2.2.1 as the 'diet' FT2 build and 2.11.1 as the latest one seems enough to me. Saved 30 KB of RAM doesn't do much benefit IMHO.

mikrosk, 2025-04-19:
It doesn't cost us anything except answering questions which one should I choose. :-) I'm always in favour to provide as few options for the normal user as possible.

th-otto, 2025-04-19:
i'm in favour to provide as many options that are possible without having more work ;)

But if it is too confusing, we could just move the extra archives  in a separate folder.  I could  also add some explanations to the download page.

mikrosk, 2025-04-19:
Yes, that's a reasonable compromise.